nrao/astrohack #13: Near Field Holography

Jan-Willem, 2023-04-20:
Add support for ALMA near-field holography.

vsmagalhaes, 2023-04-20:
* ALMA near-field holography article:
    https://ui.adsabs.harvard.edu/abs/2007IAPM...49...24B/abstract
      o All the details of the software and hardware implementation of
        the ALMA near-field holography system.  Includes a detailed
        description of the near-field correction used.
  * Robert Lehmensiek ([rlehmens@nrao.edu](mailto:rlehmens@nrao.edu)) is the NRAO antenna optics
    expert who has worked on the ngVLA optics design.  He should be able
    to answer any questions about ngVLA optics that you might have.
  * holomemo.pdf: Near Field Radio Holography of the ALMA Prototype
    Antennas (attached): Memo included in the documentation supporting
    the results from the evaluation of the ALMA prototype antennas. 
    Much of the material in this memo was included in the published
    holography article I have already pointed you to, but also includes
    some details regarding the panel/adjuster layout and numbering.
  * ALMAholo_utils.clic: This is the file used to do a number of things
    in the CLIC ALMA holography data analysis software, including noting
    the locations of BUS sectors and panels. Unfortunately all of these
    things are calculated rather than being in the form of a static
    listing, but I suspect that you can figure out how to translate this
    gildas scripting language into python.

vsmagalhaes, 2023-04-20:
[alma-holomemo.pdf](https://github.com/casangi/astrohack/files/11290341/alma-holomemo.pdf)
[ALMAholo_utils.txt](https://github.com/casangi/astrohack/files/11290343/ALMAholo_utils.txt)