nrao/astrohack #33: AstroHACK Benchmarking

Jan-Willem, 2023-05-03:
The purpose of this ticket is to develop automatic execution of performance benchmarking tests that satisfy two use-cases:
- To characterize trends in performance metrics across AstroHACK repository history (after releases are published from the main trunk).
- To detect code causing performance issues during verification of development branches (before changes are merged with the main trunk).

The implementation should be based on the work done in  [casabench](https://github.com/casangi/casabench) that makes use of [airspeed-velocity](https://asv.readthedocs.io/en/stable/). The main difference is that where casabench is a separate repository from CASA6 the AstroHACK benchmarking code can be incorporated into the same repository.

smcastro, 2023-05-04:
@Jan-Willem, is the purpose of this ticket to start the implementation of the benchmarking tests? Or only a place holder for when the process is defined for the verification and validation?

amcnicho, 2023-06-27:
We're proposing a framework that incorporates both [pytest-benchmark](https://pytest-benchmark.readthedocs.io/en/latest/) and [airspeed-velocity](https://asv.readthedocs.io/en/stable/index.html) to do performance testing of astroHACK. This will be elaborated on under the relevant section of [the wiki](https://github.com/casangi/astrohack/wiki/Test-Development#performance-tests).

closed 2023-06-30 by jrhosk