nrao/astrohack #34: Calibration Checking Function

Jan-Willem, 2023-05-03:
After the holography data has been extracted it would be useful to have a function that will extract the beam peaks (l=0,m=0) and do a couple of plots. 

Data in holog_zarr is organized as follows:
[ddi_i] [map_j] [ant_k] [time,chan,pol]

An example is given in the attached notebook:
[alma_debug_example.ipynb.zip](https://github.com/casangi/astrohack/files/11391291/alma_debug_example.ipynb.zip)

jrhosk, 2023-05-17:
Aside from naming convention in the private function : ) this is can be checked out now. If it looks good, I'll boring-ify the private functions and we can merge it. Below is a snapshot of what to expect. There are two plotting options, `matplotlib` and `plotly`, the later is interactive. I chose let the user specify the polarization because default plotting ever combination looked messy and if desired the user can just do it themselves with a quick loop.

jrhosk, 2023-05-24:
This is done and merged. Not sure why it still shows up.

closed 2023-05-24 by jrhosk