nrao/astrohack #65: API modifications

ppbbeaklini, 2023-05-17:
I am thinking about making the astrohack tasks more intuitive for casa users and more compatible with the calibration and plotting casa tasks. The primary concern is with the astrohack input dictionary; other are minor changes like using vis instead of ms_name. Below is a calibration script where I try to fit the extract_holog task using the existing variables in the CASA environment. I understand that now we are trying to handle multiple and complex things with few tasks; therefore, I made some comments to open possibilities about ways to address the difficulties.

## #Script starts here ###

msname = 'THOL0001.sb43733716.eb43740773.60029.152514375004.ms'	# Name used in calibration tasks is vis to call the input file
calfield = '1'
holofield = '2'
refant = 'ea06'
spw = '*:4~60'

gaincal(vis=msname,caltable='delay10s.cal',field=calfield,refant=refant,solint='10s',spw=spw,gaintype='K')
bandpass(vis=msname,caltable='bandpass.cal',field=calfield,refant=refant,solint='inf',spw=spw,solnorm=True,gaintable=['delay10s.cal'])
gaincal(vis=msname,caltable='gain.ap.cal',field=calfield,refant=refant,calmode='ap',solint='10s',spw=spw,gaintable=['delay10s.cal','bandpass.cal'])
applycal(vis=msname,field=holofield,spw=spw,gaintable=['delay10s.cal','bandpass.cal','gain.ap.cal'],gainfield=[calfield,calfield,calfield])

ant = 'ea27'						
plotms(vis=msname,field=holofield,spw=spw,avgchannel='56',avgtime='10',antenna= ant,xaxis='time',yaxis='amp',correlation='RR',iteraxis='spw',timerange='07:10:00~07:20:00')

###### the intuite tought to extract_holo based on what is already used #####

# 1 - Could we use "antenna" as in plotms task?
ant = 'ea27'						#plotms input (antenna = '' means all and ant = 'ea17,ea27' for two or more antennas).
# 2- refant  I do understand that refant on holo is a different variable than in calibration. 
# in AIPS, the refant in holo is called baseline, to differ from the refant input used in the calibration task
# In casa, It could be refantholo, or something like that, and the inputs could be 'ea12,ea28', or 'all', or 'nearest' or 'twonearest'
# For VLA, in most cases, we use the two nearest baselines. But perhaps we should ask ngVLA what the plan is. I will ask them.

refantholo = 'ea12' 

# 3 split task uses datacolumn to indicate 'DATA' or 'CORRECTED. I suggest we follow the same.		    

column = 'CORRECTED' # Or data. The name used in the plotms tasks is datacolumn instead of data_col

# 4 Here is a tricky one. I do understand that OTF can have different maps. Not only that but even at VLA, we can have two holo scans in the same dataset. Could the user select them based on time?

timerange = ''07:10:00~07:20:00'

holog_mds = extract_holog(vis=msname,antenna=ant,spw=spw,refant=refantholo, datacolumn='CORRECTED',parallel=True,overwrite=True,timerange=timerange,point_name=msname+'.point.zarr',holog_name=msname+'.holog.zarr')

#I wouldn't be against making a task to generate a dictionary before running holog_mds, as an example below. Opacity gencal follows a similar pattern, running first plotweather task to get the input parameters.

dic = makeholodict(vis=msname,antenna=ant,spw=spw,refant=refantholo,timerange=timerange)
holog_mds=extract_holog(vis=msname,parameter=dic,datacolumn='CORRECTED',parallel=True,overwrite=True,timerange=timerange,point_name=msname+'.point.zarr',holog_name=msname+'.holog.zarr')

ppbbeaklini, 2023-05-17:
Not sure why the items became giant letters. Sorry about that :-p

vsmagalhaes, 2023-05-19:
This issue depends on issue #72
Points to be addressed in this issue
- [ ] Standardization of parameter names across Astrohack functions
- [ ] Make parameter names closer to CASA API when possible/desirable
- [ ] Standardization of parameter types across functions
- [ ] CASA like parsing where feasible

closed 2023-06-01 by jrhosk