nrao/astrohack #99: Write unit tests for astrohack_local_client()

smcastro, 2023-06-20:
Based on the user stories defined in https://github.com/casangi/astrohack/wiki/Test-Development#unit-tests: define here the test cases that will be implemented.
```[tasklist]
### Test Cases
- [ ] test_astrohack_....
- [ ] test_astrohack_...
```

jrhosk, 2023-06-27:
Here is the code I use to verify the client has been spawned.

```
DEFAULT_DASK_ADDRESS="127.0.0.1:8786"

        logger = _get_astrohack_logger()

        if parallel:
            if not distributed.client._get_global_client():
                try:
                    distributed.Client(DEFAULT_DASK_ADDRESS, timeout=2)

                except Exception:
                    from astrohack.astrohack_client import astrohack_local_client

                    logger.info("local client not found, starting ...")

                    log_parms = {'log_level': 'DEBUG'}
                    client = astrohack_local_client(cores=2, memory_limit='8GB', log_parms=log_parms)
                    logger.info(client.dashboard_link)
```

jrhosk, 2023-07-24:
> 
> @jrhosk I'm wondering if there is a way to create another instance of the client using a different port? For example, can > each test case before creating a client set this DEFAULT_DASK_ADDRESS="127.0.0.1:8786" using a different port?

That is what it does now. That is what is meant by "Hosting the HTTP server on port 50813 instead".

> @jrhosk , should we open issues for bugs that we find in the code and documentation?

I think so, yes.

smcastro, 2023-07-25:
I see so the warning already assigns a new port to the new client.

jrhosk, 2023-07-25:
> I see so the warning already assigns a new port to the new client.

Yes, the client just opens a new port if it finds a port is open. This is honestly not ideal in my opinion, you end up with a bunch of stale, open ports. I'm working on get ports better cleaned up after use.

I have already changed the test to close the opened ports after testing in each test case.

jrhosk, 2023-07-25:
@smcastro Is there a test case for these?

- [ ] test_test_client_with_valid_args() 
- [ ] test_client_with_invalid_args()

I think all arguments should be valid. I would say there are three things to test here:
- client spawns correctly.
- `dask dir` created properly when specified.
- log file written correctly.

jrhosk, 2023-08-28:
This is finished, making pull request.

closed 2023-08-28 by jrhosk