nrao/astrohack #132: generate_holog_obs_dict ingnoring point_name input

ppbbeaklini, 2023-08-04:
It seems that, no mater which name I set, it put as default msname+point.zarr. Example:

from astrohack.extract_holog import generate_holog_obs_dict
test33 = '/lustre/aoc/sciops/pbeaklin/holography/OTF/2023-07-24/otf_holo_3c84_refnums_ms.pb.60149.74588130787.ms'
# which are the ddi inputs?

dicc =  generate_holog_obs_dict(
    ms_name = test33,
    point_name = '/lustre/aoc/sciops/pbeaklin/holography/OTF/2023-07-24/holo33x33.C.point.zarr',
    baseline_average_nearest = 2,
    )
    

Log 

2023-08-04 09:40:36,761 - ERROR    - astrohack - (_dio.py:36) - [extract_pointing]: /lustre/aoc/sciops/pbeaklin/holography/OTF/2023-07-24/otf_holo_3c84_refnums_ms.pb.60149.74588130787.point.zarr already exists. To overwite set overwrite to True, or remove current file.
2023-08-04 09:40:36,762 - ERROR    - astrohack - (extract_pointing.py:111) - There was an error, exiting:: Exception: /lustre/aoc/sciops/pbeaklin/holography/OTF/2023-07-24/otf_holo_3c84_refnums_ms.pb.60149.74588130787.point.zarr exists.

Please, notice that the point name on the log message are completely different form what I set.

jrhosk, 2023-08-04:
Set overwrite=True

ppbbeaklini, 2023-08-04:
Sure. If I set overwrite = True will work. Of course. But the point is that I want to change the name e keep both point tables. Please, noticed that the input name is different from the name that appears in the log message.

jrhosk, 2023-08-04:
Sorry, I realized after I typed it. Ithink I know what the issue is. Let me confirm and Ill explain.

jrhosk, 2023-08-04:
Alright, I have checked the code to confirm and the code is working as I expected but obviously not as you expected; it should be better documented, for that I am sorry and I will fix it.

The way the `generate_holog_obs_dict` work in regards to the `point_name` is that the parameter is assigning the pointing file, already on disk, that you want to use. It assumes you have already run `extract_pointing`. When it doesn't find the file you give it, it generated a file based on the default value of ms_name+point.zarr. So the code is work as it was coded. That said, it doesn't have to work that way, if you would prefer it be a bit more complicated, ie. `point_name`=None --> default and `point_name`="some_name.point.zarr" creates that file, it is a small change to make happen.

This needed to be better documented, for that I apologize. I will make it clear after we come to agreement on how it should work.If you want to discuss some issues on gmeet just let me know and we can iterate more quickly.

closed 2023-08-08 by jrhosk