nrao/astrohack #205: Logger seems to fail for parallel calls

vsmagalhaes, 2023-12-11:
When running the visualization notebook under python 3.8 several functions when Called in parallel throw the following messages about a hundred times:
```
Could not load worker logger: 'worker_logger'
dict_keys([])
Could not load worker logger: 'worker_logger'
dict_keys([])
Could not load worker logger: 'worker_logger'
dict_keys([])
Successful readonly open of default-locked table data/ea25_cal_small_after_fixed.split.ms/ANTENNA: 8 columns, 3 rows
Successful readonly open of default-locked table data/ea25_cal_small_after_fixed.split.ms/OBSERVATION: 9 columns, 1 rows
Could not load worker logger: 'worker_logger'
dict_keys([])
.
.
.
Could not load worker logger: 'worker_logger'
dict_keys([])
```

jrhosk, 2023-12-12:
Was this using the main branch? I couldn't replicate it earlier.

vsmagalhaes, 2023-12-12:
Yes using the main branch on python 3.8

jrhosk, 2023-12-12:
Alright, I'll take a look as soon as this test passes.

jrhosk, 2023-12-13:
The problem is in how and where the astrohack specific worker plugin is called. The logger currently has a specific place in the skriba repo where I converted this plugin but the plugin from astrohack is still called so the naming was all wrong. I have patched it for the time being and will make a more robust plugin register mechanism in the new year with the full independent logger.