nrao/astrohack #235: `extract_holog` fails to calculate `npix` for EVLA S Band holography

Kitchi, 2024-04-01:
When I attempt to run `extract_holog` on some EVLA S band holography, it fails with the following error (traceback truncated)

```
│                                                                                                  │
│ /export/home/corrino/krishna/soft/micromamba/envs/astrohack_master/lib/python3.10/site-packages/ │
│ astrohack/_utils/_extract_holog.py:122 in _extract_holog_chunk                                   │
│                                                                                                  │
│   119 │   # function more general use (hopefully). I honestly couldn't see a reason to keep it   │
│   120 │   for ant_index in vis_map_dict.keys():                                                  │
│   121 │   │   antenna_name = "_".join(("ant", ant_names[ant_index]))                             │
│ ❱ 122 │   │   n_pix, cell_size = calculate_optimal_grid_parameters(pnt_map_dict, antenna_name,   │
│   123 │   │                                                                                      │
│   124 │   │   grid_params[antenna_name] = {                                                      │
│   125 │   │   │   "n_pix": n_pix,                                                                │
│                                                                                                  │
│ /export/home/corrino/krishna/soft/micromamba/envs/astrohack_master/lib/python3.10/site-packages/ │
│ astrohack/_utils/_algorithms.py:282 in calculate_optimal_grid_parameters                         │
│                                                                                                  │
│   279 │   │    - pnt_map_dict[antenna_name].POINTING_OFFSET.values[:, 1].min())                  │
│   280 │                                                                                          │
│   281 │   try:                                                                                   │
│ ❱ 282 │   │   n_pix = int(np.ceil(data_range / cell_size)) ** 2                                  │
│   283 │                                                                                          │
│   284 │   except ZeroDivisionError as e:                                                         │
│   285 │   │   logger.error(f"Zero division error, there was likely a problem calculating the d   │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
ValueError: cannot convert float NaN to integer
```

I have put the test script in this gist - https://gist.github.com/Kitchi/c9dcd4ea1850a4955e07eb31089e9128

A minimal dataset that reproduces this issue can be found here - `/lustre/aoc/sciops/ssekhar/tests/astrohack/THOL0001_sb31146700_1.57287.317363055554_HOLORASTER.timeavg.ea28.ms`

jrhosk, 2024-04-02:
Can you change the permissions, I can't copy the file.

jrhosk, 2024-04-03:
@Kitchi This issue can be solved by setting increasing `time_smoothing_interval` in `extract_holog` from 10. --> 20. The issue comes from the time averaging where the initial pointing sample time is actually greater than the visibility sample time in the pointing table, which the way it is is now, causes there to be a divide by zero in the averaging. We are looking into whether this is something that needs a more longer-term fix but for now this fixes it on my side. I have added a loud warning in the code that yells at you when this happens for now (this will come on the development branch as long as the tests pass). Let me know if this works for you.

Also @vsmagalhaes please add more context if you would like.

Kitchi, 2024-04-03:
@jrhosk Increasing the `time_smoothing_interval` worked, thanks!

closed 2024-04-03 by jrhosk