nrao/astrohack #241: Holog fails when data is singleton and antenna and ddi are set to 'all'

vsmagalhaes, 2024-04-02:


jrhosk, 2024-04-08:
I can take a look at this if you can link an example ms.

vsmagalhaes, 2024-04-10:
Hey Josh,
There is a dataset to test this here:
https://drive.google.com/file/d/1yYesXnHjBvzI_7l01xQYb44cMoMXYN9R/view?usp=drive_link

You can get the error by calling `holog` as:
```
holog('X4.holog.zarr',
      ddi='all',
      ant='all',
      chan_average = False,
      scan_average = True,
      to_stokes = False,
      apply_mask = False,
      phase_fit = False,
      overwrite = True,
      parallel = False
      )
```

jrhosk, 2024-04-11:
This should be fixed now on the development branch. I had to change some key names in the `locit` file like `ant_info` --> `antenna_info` because the key filter was picking them up. Ultimately, I think we should have a better solutions and think a bit about how and what we add to the mds files. Maybe we could just add a key like "meta_data" and the everything like 'ant_info' would appear under that. This would also allow us to standardize the mds formats a bit more.

Test it out and I'll close the issue when you are happy.

closed 2024-04-11 by vsmagalhaes