nrao/astrohack #260: Create a user friendly way to access holog obs dict after running of extract_holog

vsmagalhaes, 2024-08-15:


jrhosk, 2024-08-15:
There is already a function for this though. If the user wants a copy they should create it before running extract_holog or load it from disk, right?

vsmagalhaes, 2024-08-15:
Yeah, But pedro wants to see which baselines were used after extract_holog was run without giving a holog_obs_dict.
For that we need to access the holog_obs_dict created by extract_holog.
It is saved on the hidden .json file which is not the most user-friendly way of retrieving this.

jrhosk, 2024-08-15:
but there is a holog obs dict object, with a collapsible json views AND a selection function that I built especially for him and he knows about because I personallyl showed it to him and made a tutorial notebook with it...

vsmagalhaes, 2024-08-15:
Yeah, but how do you read it from disk?
And the .josn is saved at the execution directory, not inside the .holog.zarr which means that you loose it if you run extract_holog multiple times.
My idea is to save .holog_obs_dict.json inside the .holog.zarr and to return its associated object with a method of the holog_mds object.
Just simplify its access and avoid it being overwritten

vsmagalhaes, 2024-08-15:
I am here with Pedro doing a step by step reduction that is why I am spamming new issues

jrhosk, 2024-08-15:
You just instantiate the object with the file directory as an argument, like loading any other file. I would be behind moving the file directory, though that will make things confusing when you need to generate one maybe but making another method to return and object with more methods, I really thing we should avoid this bloat. The summary() function is already overburdened on some files. All the users has to do is load the dictionary into the object and they have a readable, collapsible object (it is even search able iirc).

vsmagalhaes, 2024-08-16:
Hey Josh,
These changes sound good, thanks for doing this.
We can iterate next week about a way to put the .holog_obs_dict inside the .holog.zarr and how to deal with it when we generate it ad hoc with generate_holog_obs_dict.