nrao/astrohack #278: Antenna exclusion in baseline selection

vsmagalhaes, 2024-10-25:
Sometimes the closest antenna to a mapping antenna has bad data in these situations it is useful to exclude antennas from the selection of baselines for the mapping antennas.

ppbbeaklini, 2024-10-28:
Thank you for opening the ticket, Victor.
That happens when the shortest baseline antenna is noisy, and I want to use the second shortest baseline, no matter what the antenna is. That is similar to what the casa pipeline has, which is getting the shortest baseline by ignoring antenna eaxx.

jrhosk, 2024-10-28:
Is this something that needs to be done in `HologObsDict(...)`? If it is the selection from there, it is an easy enough fix to allow filtering with an exclude list.

vsmagalhaes, 2024-10-28:
@jrhosk It is exactly that, I am already working on it.
My idea is to simply jump over to the next antenna if an antenna shows up on an exclusion list.

jrhosk, 2024-10-28:
That should be fine, if you add the exclusion option to the instantiation you could do the checks in first pass so you can optimize the looping. Sounds like you got it covered, I'm out with sickness most of the week anyway.

closed 2024-11-04 by vsmagalhaes